imfatant/test #1: Setup BBBlue with Ardurover - Write error APHW

bj-neilson, 2018-05-09:
Hi,  I'm learning my way with the BBBlue and Ardupilot...slowly.  Thank you so much for the terrific installation guide.  It's great.  I'm just having trouble at the end as per the attached screen shot.  

/bin/echo: write error: Operation not permitted
/usr/bin/ardupilot/aphw: line 6: /sys/class/gpio/gpio80/direction: No such file or directory
/usr/bin/ardupilot/aphw: line 7: /sys/class/gpio/gpio80/value: No such file or directory
Started ArduRover Service.
RCOutputAioPRU.cpp:SIGBUS error gernerated
scan:669: scan past end of eeprom
ardurover.service: Main process exited, code=exited, status=1/FAILURE
ardurover.service: Unit entered failed state.
ardurover.service: Failed with result 'exit-code'.
[image]
Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6f1a000

I'd really appreciate a hand to sort it out.  So so close to having things working! Thanks again.

bj-neilson, 2018-05-09:
Update...seemed to have found more or less the solution.  From wolfeidau/setup.md...
Used his Ardurover.service file, doing away with aphw and /etc/default/ardupilot that are referenced in the Infatant solution.

[Unit]
Description=Ardurover Service
After=bb-wl18xx-wlan0.service

[Service]
ExecStartPre=/bin/echo uart > /sys/devices/platform/ocp/ocp\:P9_21_pinmux/state
ExecStartPre=/bin/echo uart > /sys/devices/platform/ocp/ocp\:P9_22_pinmux/state
ExecStart=/usr/bin/ardupilot/blue-ardurover -A udp:192.168.0.103:14550 -B /dev/ttyS2
StandardOutput=null

[Install]
WantedBy=multi-user.target
Alias=Ardurover.service

Now connected by wifi to ArduPilot.  Just figuring out why no GPS and telem radio.  Thanks guys.

fatdollar, 2018-05-11:
I'm having this same issue.
I don't fully understand the changes you made @bj-neilson.  I've also looked and the 2 "missing files" are there so I don't know why it says they are not.
[image]
So I'm wondering what to do at this point.  Thanks.

imfatant, 2018-05-14:
This issue will occur if you use a later Debian Stretch image than the one I recommend at the beginning of the guide. The images using the 4_14 kernel are not stable.

fatdollar, 2018-05-14:
That is inaccurate as I have always been using the 4_9 kernel.

bj-neilson, 2018-05-14:
Thanks for the feedback @imfatant .  Hope you’ve got yours going @fatdollar . 
I had an enforced break from BBBl after toasting one. New one arrived yesterday so I’ll try again!

imfatant, 2018-05-15:
Did you use the image I recommend? (And sorry about the late reply, I've been busy the last week.)

https://rcn-ee.net/rootfs/bb.org/testing/2018-04-22/stretch-console/bone-debian-9.4-console-armhf-2018-04-22-1gb.img.xz

imfatant, 2018-05-15:
OK, I just fresh re-installed to check, and the steps as shown in the guide work perfectly. I assume that you have made the ArduPilot program files (arducopter, arduplane, etc) executable by doing something like `sudo chmod 0744 /usr/bin/ardupilot/ardu* `, not that this is the problem you indicate.

I can only suggest you *really do* use the specific Debian image I recommend:

https://rcn-ee.net/rootfs/bb.org/testing/2018-04-22/stretch-console/bone-debian-9.4-console-armhf-2018-04-22-1gb.img.xz

Do not use a later one.

Note that you should use a specific ardurover service file, if you are indeed using ardurover. I have added one to the guide. The aphw file doesn't change. It is used by all the flavours of ArduPilot.

The problem you both identify occurs when one or more of the commands in the aphw file fails. When this happens, the aphw file fails to return success to the service file that pre-executes it. You can get around this by removing the line `ExecStartPre=/usr/bin/ardupilot/aphw` in the service files.

But you'll still need to switch on power to the +5V servo rail:
/bin/echo 80 >/sys/class/gpio/export
/bin/echo out >/sys/class/gpio/gpio80/direction
/bin/echo 1 >/sys/class/gpio/gpio80/value

And enable the PRU (the 4th line):
/bin/echo pruecapin_pu >/sys/devices/platform/ocp/ocp:P8_15_pinmux/state
Let me know how you get on.

bj-neilson, 2018-05-15:
Thanks @imfatant. I did a fresh install yesterday on a new BBBlue. Worked 1st time. Don’t know the difference as I used that image both times. Did use chmod 0744. Thanks for the added detail though...learning Linux on the run with this so every extra bit of explanation is brilliant!

Waiting not so patiently for some connectors for my Linear actuators to arrive in the mail.  Can’t wait to get a vehicle moving!

Thanks again for the help 😁👍

imfatant, 2018-05-15:
Hehe. Learning Linux and doing this at the same time is quite a test. A good way to learn Linux, though. And of course, I'm going to recommend Arch Linux, which has an awesome forum + community if you seriously want to learn (more generally).

Of course we use Debian for the BBBlue at the moment.

bj-neilson, 2018-05-15:
@imfatant Found myself with a little free time for the first time in a while 😩. I’ll have a look at Arch.  Good to know of support!   Got Ubuntu installed on a spare laptop at the moment but been using Win10 mostly so far.  Been a change...largely used Macs for years now.  Steep learning curve! Long time since uni Fortean 77, SQL etc.  I’m thinking ROS in a couple of weeks....like a challenge 😁.  (Sorry for the off topic)

imfatant, 2018-05-15:
S'all good :)

bj-neilson, 2018-05-17:
To hopefully add something to this string....I’d been using a brand new HP with Win10 to try the installation.  Never quite got there.  Tried doing it exactly as per your guide this morning on the same machine via an Ubuntu installation and it worked perfectly.  A bit of cutting and soldering later and I had udp and radio telemetry (433Mhz hobby king radios) working and also gps (Ublox neo  m8n). The only bother was trying to install a GCS on Ubuntu.  Didn’t get there.  Used Win10 for Mission Planner.

imfatant, 2018-05-17:
Depending on how you've 'dualed' your computer with the operating systems, it may be that the router's DHCP server is handing it different IPs. Check this by running something like nmap on both OSes.

Just a thought.

bj-neilson, 2018-05-17:
Cheers for the thoughts. Na didn’t really get that far. Installed package but the link to the desktop file was broken and I was running short of time. I’ll have another look. One more a actuator to mount on my machine, electronics to mount in enclosure and I’ve about got a start 👍. Need a few rainy days so I can work in the shed.
[image]

fatdollar, 2018-05-23:
Mine is running but I am still seeing this same error but doesn't seem to affect the function of the autopilot.  I have not flown yet but communication with Mission Planner is great.  I just need to connect the motors now and make sure they are spinning in the correct orientation and I will see how it flies.

imfatant, 2018-05-24:
Glad it's running, but if you're following the guide (with the image I recommend), I can't see why you're getting an error. That is baffling.

bj-neilson, 2018-05-24:
I’ll check mine tomorrow when I get a moment.  As I’d mentioned above all of my little issues were resolved once I followed the guide and did it via an Ubuntu laptop. Running so beautifully I didn’t check the log files 😁

imfatant, 2018-05-26:
I left a note @https://gitter.im/mirkix/BBBMINI regarding the 'scan:681: scan past end of eeprom' message.

closed 2018-06-19 by imfatant