imfatant/test #3: RE: Part 2 #16/Compiling on the BBBlue

silver2row, 2018-05-23:
Hello,

I cannot compile on my BBBlue for some reason. I know it will take forever and I should use the Ubuntu machine like last time. I would rather try to compile on the BBBlue this time instead of Cross-Compiling. 

I am having issues on Part 2 #16...

Should I use sudo chmod 0755 xx/xx/xx/x on my file(s) or should I just try to compile them as is?

Also, I am assuming I need to get at a point when I can use ./waf to compile my software on the BBBlue but I have been unable to get info. on ./waf at all.

Seth 

P.S. If this makes any sense, please reply. I sure could use some pointers.

silver2row, 2018-05-23:
Hello Once More,

I tried to get "waf" on my system, i.e. the BBBlue w/ the suggested Linux distro from your "test" page.

Do you know how I can get this waf compiler install on my system?

Seth

silver2row, 2018-05-24:
Hello,

I found that sometimes the u-boot overlay section on the Linux uEnv.txt file should be managed, e.g. w/ it commented-out and kernel 4.9.x typed into that location.

I found this idea via an online search. I am still getting the error b/c my uEnv.txt file was already set up to suit the needs of this config.

Seth

P.S. Any help would be good on this error file. I looked online under some software at github.com and found that the end word, generated, is actually spelled out gernerated on their software.

imfatant, 2018-05-24:
@silver2row 

The subject line of your issue is regarding compiling ArduPilot on the BBBlue. I'm not sure how that relates to your actual question. If you have any questions on that matter, please let me know.

Your actual question is regarding the RCOutputAioPRU.cpp:SIGBUS error. This error arises when you attempt to use a 4_14 image, which I emphasize in the guide, must not be done. You have noticed that the 4_14 kernel is specified in uEnv.txt. This is because this is a 4_14 image.

Do not do this. Use the image I recommend. See Step 1 of the guide.

I hope this helps.

-- Imf

silver2row, 2018-05-24:
Hello,

Okay. I thought I was using the 4.9 kernel but I will check again. I will get back to you.

Seth

silver2row, 2018-05-24:
@imfatant ...

Hello...I have used this cmd: uname -r.

    4.9.88-ti-rt-r111

Seth

P.S. That is the one I am using. If you have tested it and know what I may be doing incorrectly, please let me know. I am at a loss right now.

silver2row, 2018-05-25:
Hello...I am going to try some other images and I will return service.

Seth

imfatant, 2018-05-26:
A point of clarification: uname -r shows you the kernel version you're using, not the image version. Images later than 2018-04-22 are designed to use the 4_14 kernel (which you can tell by looking at the uncommented 4_14 line in uEnv.txt). By all means try them, and use the 4_14 kernel, but they aren't stable (at least, they aren't from my testing).

silver2row, 2018-05-26:
Hello @imfatant , 

I found out what I had been missing out of a couple of hyperlinks and some instructions on those links. 

Anyway, thank you for posting this info. and redirecting other people, including me, to repos that have other info. for starters like me.

Seth

P.S. I appreciate what you have done. I will reply once more after some further testing has been completed. If you are tired of this back and forth, we can end it. Just reply and i can close this issue if this is what you want.

silver2row, 2018-05-26:
@imfatant 

Do you know anything about these issues?

    Sensor failure: Baro: unable to initialise driver
    Segmentation fault

and...

    Scheduler: failed to set scheduling parameters: Operation not permitted

I will look through the issues online and via ardupilot on GitHub.com and on their site.

Seth

P.S. Anyway, thank you so far.

ejhaze, 2018-08-08:
What were the fixes for this? 

Here is the error and the uname output: 

RCOutputAioPRU.cpp:SIGBUS error generated
root@beaglebone:/usr/bin/ardupilot# uname -r
4.4.113-ti-rt-r149

I started with a fresh reflashed BBB and still ran into this. 

At a certain point following mirkix, bbbmini and downloading a whole bunch of python files, I did have the Blue working with Mission Planner. I just couldn't get it to recognize any pins at the time. If I'm not mistaken, he was using 4.9.

ejhaze, 2018-08-08:
That's what I'm saying, I started from scratch, reflashed with the Blue, followed this guide, and got the error. I don't see

imfatant, 2018-08-08:
This bit ???

---+++ The real-time kernel is currently broken! Use the following fix: +++---

Specify Ti real-time kernel 4_4. Do NOT use 4_14: sudo /opt/scripts/tools/update_kernel.sh --ti-rt-channel --lts-4_4

sudo sed -i 's|uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo|#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo|g' /boot/uEnv.txt

sudo sed -i 's|#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo|uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo|g' /boot/uEnv.txt

Finally, copy am335x-boneblue.dtb (from the bin directory in this repo) to /boot/dtbs/4.4.113-ti-rt-r149

ejhaze, 2018-08-09:
Ran it a second time today and it worked. I suppose the the scp/ftp of the am335x-boneblue.dtb must not have transferred correctly. That would be the only thing different. Thanks!

closed 2018-08-09 by imfatant