imfatant/test #4: RE: Info. and Ideas for ArduCopter-3.5.5/BeagleBone Blue...

silver2row, 2018-05-27:
Hello,

    Init APM:Copter V3.5.5 (27229c83)

    Free RAM: 262144
    FW Ver: 120
    ----------------------------------------

    Sensor failure: Baro: unable to initialise driver
    Segmentation fault

I just ran my arducopter software for the first time, i.e. sort of. Anyway, I received this errror.

and...

I know that I may have to use a real IP address instead of what the BBB.io people told us to use, e.g. 192.168.7.2. I am typing this b/c the 192.168.7.2 is what I have been using to sign on for years now.

Now...

Should I recompile my arducopter version from 3.5.5 to another version or should I look to another source for what could be the issue?

Seth

P.S. If you have dealt w/ this info. from above, please allow me some input. I will keep plugging at it. Thank you for any insight.

silver2row, 2018-05-27:
Hello...I am reverting back to arducopter-3_5_4 like you have stated in your info. on /test.

Also...

    lttng-ust          not found
    libiio                not found
    benchmarks     disabled

This info. is some of the info. that stood out to me after the cmd of **sudo ./waf --board=blue**

I tried a new IP address too.

Seth

P.S. If you have any ideas as to what may be causing my confusion on this subject, please chime in. Thank you.

silver2row, 2018-05-27:
Hello Once More,

Look...I know it may seem like I am bothering you. I am sorry for that idea but I have some viable concerns and no outlets/resources. Please view what I have come across w/ 3.5.4 instead of 3.5.5.

    Sensor failure: Baro: unable to initialise driver
    Segmentation fault

I cannot get the system to start. Instead, it provides these errors codes. I am off to look up these issues.

Seth

P.S. I will update you further in the future.

imfatant, 2018-05-27:
Hi Seth,

The 'Sensor failure: Baro: unable to initialise driver' error is weird. Occasionally people get dud boards.

That is not a v3.5.5 or a v3.5.4 ArduCopter issue. I've never seen it myself. You could ask on gitter? I vaguely recall someone having a similar thing happen with the IMU seg faulting.

By the way, I'm running v3.5.5 at the moment. It's fine. I believe that's what's shown in the guide.

The compilation method I outline in the guide does work. I checked yesterday. I'm afraid those errors are  very strange..

- Imf

silver2row, 2018-05-28:
Hello,

No issue. I will try to plug at it some more.

Seth

closed 2018-05-28 by silver2row