imfatant/test #5: RE: Hello and Setting Up ArduCopter/Seth

silver2row, 2018-07-14:
Hello @imfatant ,

Seth here. I followed the directions on your page to the "T." I left a message previously but I decided to take a break for a bit. I am back on w/ ArduPilot and trying again to make things work w/ my BBBlue.

Anyway, here is the software failure when I run the cmd: sudo /usr/bin/ardupilot/arducopter -A udp:<my IP Address>:14550.

    RCOutputAioPRU.cpp:SIGBUS error gernerated

Is this a normal error or am I entering in the cmd to run arducopter incorrectly?

Seth

P.S. When you have an extra minute or two, will you please let me know what this exactly means?

imfatant, 2018-07-15:
Hi, welcome back :)

This has been a topic of discussion on the gitter channel last few days, so please have a look there. I'm currently trying a few things out, but it looks like the RT kernel is broken..

silver2row, 2018-07-15:
Okay...

I have found that when I use:

    sudo ./waf configure --board=blue

...

I get lttng-ust not found and libiio not found. Also, when I perform:

    sudo ./waf

I receive some notice of:

XML Validator Libraries not found and not using XML Validation.

Seth

P.S. If this is of any importance, please attach it to other ideas. I will keep trying. I have tried to test w/ other forms of the ArduPilot/ardupilot w/:

    sudo git checkout Copter-3.5.7-rc1

...

I have not been successful w/ getting updated versions.

imfatant, 2018-07-15:
> Hello Again and You are Free to Close When You Get Bored w/ Me,
>
> @imfatant <https://github.com/imfatant> , I have been researching the
> lost libraries that the system states I do not carry. IIOD and its
> dependencies, lttng-ust which can be found here at
> https://lttng.org/docs/v2.9/#doc-debian, and the XML Validation Libraries
> that were not allocated to the compilation when I compiled w/:
>
> sudo ./waf
>
> ...
>
> If you have time to review any of it and if you think those ideas matter,
> please reply.
>
> Seth

Seth,

The current fix is to do this:
  1) sudo /opt/scripts/tools/update_kernel.sh --ti-rt-channel --lts-4_4
  2) Copy https://files.gitter.im/mirkix/BBBMINI/Teqp/am335x-boneblue.dtb
to the /boot/dtbs/4.4.113-ti-rt-r149 folder.

That should get rid of the `RCOutputAioPRU.cpp:SIGBUS error gernerated`
message.

silver2row, 2018-07-15:
Hello,

Thank you. I am off to test it.

Seth

silver2row, 2018-07-16:
Okay Sir,

I recompiled after erasing ardupilot. I got the file you have listed and put that file in the place you told me to put it, i.e. /boot/dtbs/4.4.113-ti-rt-r149.

The board looks like it is running w/ the user LED working in Red but I forgot to copy the ./build/blue/bin/a* into /usr/bin/ardupilot. 

Now, I am stuck. I rebooted the BBBlue and got an error when trying to cp the ./build/blue/bin/a* into the new directory.

    cp: cannot create regular file '/usr/bin/ardupilot/arducopter': Text file busy

Seth

P.S. What do you think, at this point, I can do or should I start over and recompile again?

silver2row, 2018-07-16:
@imfatant ,

Sir...look here:

    Sensor failure: Baro: unable to initialise driver
    Segmentation fault

This was my error after I started the .service file for arducopter and ran the udp:xxx.xxx.xxxx.xx:14550.

Seth

silver2row, 2018-07-16:
Hello,

This is my set up for now. It runs but does not run as expected. 

    setup:307: bad header in setup - erasing
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver

Seth

silver2row, 2018-07-16:
Hello Again Sir,

Thank you for your help and assistance and the awesome tutorials.

Seth

closed 2018-07-16 by silver2row