imfatant/test #11: Battery Voltage Not being reported.

fatdollar, 2018-11-01:
The battery voltage being reported by my system is 0V using both Mission Planner and QGroundControl.  I have tried different settings within both of those control stations but I am always getting 0V. Is there a setting on the board that needs to change?
Have you come across this issue?
I con't really want to fly this until I can safely monitor my battery levels.  In addition it doesn't pass pre-flight checks anyways unless I override the battery check.

Any ideas?

imfatant, 2018-11-04:
Hi,

Some info about ArduPilot voltage & current sensing:

An0 is for voltage and An1 is for current. You'll need a voltage divider (multiturn potentiometer) to get the voltage down to a MAXIMUM of 1.8 V on the two ADC pins (0 and 1).

The ADCs are accessible via the linux file system.
To read ADC 0: cat /sys/bus/iio/devices/iio:device0/in_voltage0_raw
To read ADC 0: cat /sys/bus/iio/devices/iio:device0/in_voltage1_raw
It's a 12-bit converter, so you read from 0 to 4096.

Here's how the battery monitor is defined: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BattMonitor/AP_BattMonitor_Analog.h#L66

Rather than use a voltage divider, the best option would be to make a little op amp circuit - offset and gain to have maximum span (1.8 V span compared to 300-400 mV span with a divider).

Hope this helps.

-- Anthony

closed 2018-11-11 by imfatant

dorovl, 2019-09-20:
Hello,

I configured the battery voltage indicator to work without the need for any additional hardware like a voltage divider. Indeed, if you look at the BeagleBone Blue Rev. A2 schematics below, **AIN6** already measures the LiPo battery voltage through a voltage divider, while **AIN5** measures the DC Jack voltage (also through a voltage divider):

[image]

Please note that I'm using ArduCopter V.3.6.10.

Here are the steps to configure using QGroundControl (daily build from 19 sept. 2019), but similar things might also be done with APMPlanner2 or Mission Planner:

Connect using QGroundControl.

Set BATT_MONITOR to "Analog Voltage Only" and save:

[image]

Close QGroundControl.

Restart the arducopter (for example arducopter-heli):

#sudo systemctl restart arducopter-heli.service

Relaunch QGroundControl.

Set BATT_CURR_PIN to Disabled and save:

[image]

Set BATT_VOLT_PIN to 6 as "Manual Entry" (6 for AIN6) and save:

[image]

After this, you should see an approximate indication of the battery voltage.

Set BATT_VOLT_MULT to 11.037:

[image]

Note: 11.037 is a value specific to my hardware and is provided hare as an indication. This value should be calibrated according to your hardware-specific voltage divider resistors ratio, by measuring the battery voltage with a multi-meter.

You might also set the "BATT_CRT_VOLT" at around 6.2V, "BATT_LOW_VOLT" at around 6.4V and "BATT_CAPACITY" according to you battery capacity. Finally, set the "Minimum arming voltage" under the Power Setup Menu:

[image]

It should also be noted the "Power Setup Menu" provides a very convenient calculator for the voltage calibration.

Hope this helps.

-- Vlad

imfatant, 2019-09-21:
Hi Vlad,

That's excellent work!

Do you mind if I add this to the guide?

Best regards,

-- Anthony

dorovl, 2019-09-22:
Hi Anthony,

Surely you should add-it in your excellent guide! You made an amazing job by collecting all this valuable information! Thanks to your work, I was able to compile the flight controller in an evening. I still need to get the necessary hardware, but I am very impatient to install it on my own helicopter (an old Raptor 30 v2).

Here are three topics from my own experience that you might also add to the guide, if you think necessary:

1. After changing the services files (as indicated in you guide), I has some strange behaviors. Therefore, I recommend always make a:

#sudo systemctl daemon-reload

every time a service file is added or edited, to make sure the systemd manager configuration is reload.

2) The battery indicator LEDs (BATT_LED_1, BATT_LED_2, BATT_LED_3, BATT_LED_4) are very useful because they visually report the battery charging status and also the voltage level:

[image]

The Robot Control Library (https://github.com/StrawsonDesign/librobotcontrol) includes a nice service, named "rc_battery_monitor" at https://github.com/StrawsonDesign/librobotcontrol/tree/master/services/rc_battery_monitor

Note: In https://github.com/StrawsonDesign/librobotcontrol/blob/master/library/src/io/adc.c, the V_DIV_RATIO is set to 11.0. For a more precise calibration this value might be configurable by the user.

Below are the instructions on how to install the rc_battery_monitor service (derived from http://strawsondesign.com/docs/librobotcontrol/installation.html):

#sudo apt update && sudo apt upgrade librobotcontrol

To the question "_Which program to run on boot?", answer "none".

After the installation, the service is automatically enabled and active:

The BATT_LED_1, BATT_LED_2, BATT_LED_3 and BATT_LED_4 leds should now be active. 

Run this command to check the service is running:
#systemctl status rc_battery_monitor.service

The robotcontrol service should be stopped and disabled (this service fails to start because module 'pru_rproc' and 'pruss_soc_bus' are missing - probably because of a wrong kernel version):

#sudo systemctl stop robotcontrol.service
#sudo systemctl disable robotcontrol.service

You might also run the rc_battery_monitor interactively. For this:

#sudo systemctl stop rc_battery_monitor.service
#sudo /usr/bin/rc_battery_monitor

2S Pack   Jack   #Cells   Cell
 8.15V   12.18V  2       4.08V

After that, don't forget to start the service back by issuing a:

#sudo systemctl start rc_battery_monitor.service

3) Finally, here are pair of utility scripts to watch the voltage on the AINs:

#cat adc1.sh
#!/bin/bash
while true; do cat /sys/bus/iio/devices/iio:device0/in_voltage* | paste - - - - - - - - ; sleep 1; done

and

#cat adc2.sh
#!/bin/bash
watch -n 1 -td 'echo "AIN0\tAIN1\tAIN2\tAIN3\tAIN4\tAIN5\tAIN6\tAIN7"; cat /sys/bus/iio/devices/iio:device0/in_voltage* | paste - - - - - - - - '

These scripts might be useful during the test of an airspeed meter sensor or other device providing an analog signal.

Best regards,
--Vlad

dorovl, 2019-09-30:
Hi Antony,

I wanted to play with the Robot Control Library mentioned in my previous post. This library provides a lot of nice examples and diagnosis tools located in /usr/bin/rc_* like:

```
rc_altitude           rc_calibrate_gyro   rc_test_adc                    rc_test_encoders       rc_test_motors
rc_balance            rc_calibrate_mag    rc_test_algebra                rc_test_encoders_eqep  rc_test_mpu
rc_battery_monitor    rc_check_battery    rc_test_bmp                    rc_test_encoders_pru   rc_test_polynomial
rc_benchmark_algebra  rc_cpu              rc_test_buttons                rc_test_escs           rc_test_pthread
rc_bind_dsm           rc_dsm_passthrough  rc_test_complementary_filters  rc_test_filters        rc_test_servos
rc_blink              rc_kill             rc_test_dmp                    rc_test_kalman         rc_test_time
rc_calibrate_accel    rc_model            rc_test_dmp_tap                rc_test_leds           rc_test_vector
rc_calibrate_dsm      rc_spi_loopback     rc_test_drivers                rc_test_matrix         rc_uart_loopback
rc_calibrate_escs     rc_startup_routine  rc_test_dsm                    rc_test_mavlink        rc_version
```

The documentation is here: http://strawsondesign.com/docs/librobotcontrol/examples.html

Robot Control Library requires a new way of communicating with the on-board PRU's, known as RPROC, instead of the UIO (Linux Userspace I/O interface) used by ArduPilot. In the past, UIO needed a 'bone' version of the kernel, while the 'rproc' a 'TI' kernel version (mutually exclusive). Now, the latest TI kernels are compatible with both versions, however the "uboot_overlay_pru" option of "/boot/uEnv.txt" must be set accordingly.

For this:

`sudo /opt/scripts/tools/update_kernel.sh --lts-4_19 --ti-rt-channel`

Tweak /boot/uEnv.txt like :

```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
```
Then:

`sudo reboot`

```
rc_test_drivers

Kernel: 4.19.73-ti-rt-r27
BeagleBoard.org Debian Image 2019-09-08
Debian: 9.11

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.4
```

Once you played with examples from "/usr/bin/rc_*", tweak back the "/boot/uEnv.txt" for ArduPilot : 

```
###pru_rproc (4.19.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
```

`sudo reboot`

**WARNINGS:** 
- I don't tested the **--lts-4_19 --ti-rt-channel** kernel with ArduPilot, despite it seems to work (I'm still waiting for some hardware). Maybe you can do some fly tests and update the guide?
- If you have problems with the ti-rt-channel kernel, then roll back with:
```
sudo /opt/scripts/tools/update_kernel.sh --lts-4_19 --bone-rt-channel
sudo reboot
```
References:
- http://strawsondesign.com/docs/librobotcontrol/installation.html
- https://gist.github.com/jonlidgard/1d9e0e92b4f219f3f40edfed260b851e

Hope this helps.
-- Vlad

imfatant, 2019-10-02:
That's superb!

Life is a bit busy this week, but I hope to look at all you've done on the
weekend. Looking forward to it:)

-- Anthony

dorovl, 2019-10-02:
No problem!🙂 I suspect that the TI RT kernel would be less efficient than the bone RT.

htop with bone RT kernel:
[image]

htop with TI RT kernel:
[image]