imfatant/test #16: RCOutputAioPRU.cpp:SIGBUS error generated

Knaldgas, 2019-03-28:
Hi @imfatant 
Thanks for providing this guide, it's exactly what we need.
I have BeagleBone Blue, flashed it with the [image](https://rcn-ee.net/rootfs/bb.org/testing/2019-03-17/stretch-console/bone-debian-9.8-console-armhf-2019-03-17-1gb.img.xz) linked on the guide.
I have updated the kernel: 
> /opt/scripts/tools/update_kernel.sh --lts-4_19 --bone-rt-channel

Besides some complaining about locale and falling back to "C", all seems good.

I have compiled and installed ardupilot as described (build on the board, took a few hours :)).
Running every command in /usr/bin/ardupilot/aphw works fine.
Running ardurover (as root) fails with:
```
scan:688: scan past end of eeprom
scan:688: scan past end of eeprom
RCOutputAioPRU.cpp:SIGBUS error generated
```
Running any other, e.g. arduplane (as root) fails with:
```
RCOutputAioPRU.cpp:SIGBUS error generated
```

dmesg returns the following after each attempt:
```
[ 1345.051433] Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6fe4000
[ 1345.051449] pgd = 3e454bc4
[ 1345.051453] [b6fe4000] *pgd=9ae23831, *pte=4a324343, *ppte=4a324833
```

Right now the service is disabled, and I'm running the commands manually. I've not yet started on Part 3 (peripherals), as the service didn't want to start.

I've read through the other issues, and searched google, but haven't found a solution yet.

I've been through the guide a couple of times, and I can't see what I've missed.
Any hints?

~Per

imfatant, 2019-03-28:
There are a couple of things to try, and I'll be updating the guide accordingly tonight or tomorrow. Here's a preview:

1) In /boot/uEnv.txt:
Ensure that the line `uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo` is commented out thus: `#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo`
Next, uncomment the line `#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo` so that it reads `uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo`.
Then, type `lsmod | grep pru`
Hopefully, you'll see:
uio_pruss              16384  0
uio                    20480  2 uio_pruss,uio_pdrv_genirq

2) There seems to be an issue with the onboard eMMC, perhaps because people are trying to burn too large an image onto it (although the image you create by STRICTLY following the guide should fit). Nevertheless, for now, you can force the BBBlue always to boot from the SD card, but we're going to do this by intentionally corrupting the boot sector of the eMMC.
***BACK UP ANY IMPORTANT DATA ON THE eMMC BEFORE DOING THIS***
`sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10`

Hope these steps solve your problem.

-- Imf

Knaldgas, 2019-03-29:
Thanks! - Step 1 was exactly the issue - ardurover now starts.
It still writes:
```
scan:688: scan past end of eeprom
scan:688: scan past end of eeprom

```
but it starts :)

Can we safely ignore those two complaints?
We're running without SDCard - Is that ok?

~Per

imfatant, 2019-03-29:
Yes, that's all normal. You can run with or without the SD Card. The important thing NOT to do is:
1. Boot off the SD Card.
2. Peform the steps in the guide (including expanding the partition with `/opt/scripts/tools/grow_partition.sh`).
3. Install lots of extra stuff onto this partition.
4. Try to flash the lot to the eMMC by uncommenting `cmdline=init=/g'` in /boot/uEnv.txt.
If there's too much data, this appears to cause a problem. I don't have the time or inclination right now to find out exactly why...
:)

Knaldgas, 2019-03-29:
Until now I have always flashed the eMMC directly from the downloaded image, and then started updating directly on the eMMC, so we are fine - Thanks! :-)

closed 2019-03-29 by Knaldgas

swissembedded, 2019-09-08:
for me it has been 1) and 2), had to add entry in /etc/modules to load uio_pruss and erase flash.