imfatant/test #17: Using DSM2 port with Lemon DSM2RX receiver.

tiago1405, 2019-04-03:
I am trying to setup and calibrate the radio controller using a Lemon DSM2RX, Blade MLP4DSM Controller, and QGroundControl (V.3.5.0). The ground control detects the BeagleBone running Arducopter, however when calibrating the controller it doesnt seem to detect my controller. I made the "-C" option in the /etc/default/ardupilot set to "ttyO4" for the DSM2 (UART4) slot on the beagle. Is there anything I'm missing?

tiago1405, 2019-04-03:
The contents of my /etc/default/ardupilot file:

TELEM1="-C /dev/ttyS4"
TELEM2="-A udp:10.1.192.219:14550"
GPS="-B /dev/ttyS2"

imfatant, 2019-04-04:
Hi,

You connect DSM to connector E4 pin 4, not UART4.

Hope that helps.

-- Imf

imfatant, 2019-04-04:
Hang on.. let me check. You might be right.

imfatant, 2019-04-04:
OK, sorry about my confusion. I use SBUS on Pin 4 of connector E4. I have not tried the DSM UART.

TELEM1 does not do what you think it does. Its supplies (outputs) telemetry data. The R/C receiver is nothing to do with the /etc/default/ardupilot file.

The problem is that I only have experience of SBUS.

tiago1405, 2019-04-04:
Ok, so would I need to do any setup in the Linux for connector E4 pin 4? I tried to rig that up using power and ground from the pwr connector and pin4 from E4 directly into my Lemon DSM2Rx.
Is there also any way to just control the drone through the ground station software? (I am going to a hobby store later today to look at more receivers so if you have any suggestions it would be greatly appreciated!)

imfatant, 2019-04-04:
Hi,

You should try both SBUS pins (to see if one works). These are marked on the pinout diagram.

There is no Linux software setup required for this.

Yes, there is a way to control the drone through the GC software - you can use manual mode, or various autopilot modes. You should begin by reading (all) the ardupilot documentation on ardupilot.org.

FrSky gear is good, and you can combine this with modules like Crossfire. But don't give up on your Lemob gear. There's a great deal of information out there, and it takes quite a bit of time to absorb.

tiago1405, 2019-04-04:
So, I got a new receiver which is a Spektrum SPMAR410 DSM2/DSMX 4CH, and I got it powered and binded to the remote. Should I be good to power up the ESC and control it now? (QGC still doesn't seem to be able to complete radio calibration for some reason).

imfatant, 2019-04-05:
You need to get the radio working first before you play around with the
ESC's and motors. By the way, the Rx pin on the DSM UART is the one I'd
recommend you try first.

closed 2019-04-16 by imfatant