imfatant/test #18: No Servo Output?

PaddyMcDonnell, 2019-05-25:
Hi,

I have followed this guide exactly (well using the closest image that is available), worked through some of the issues that other people have been having and eventually got to the point where everything seems to be running ok but alas there does not seem to be any output to the servos.

I am trying to get this going on some rovers for my sons, have tried three different BBB's and many different attempts but always the same result. Have compiled code, used linked code, tried testing with plane as recommended but they all show output (eg in raw sensor view) but nothing happens. Of course if I bypass the BBB and connect the receiver straight to the motor driver everything works as expected. I also know that the BBB works because I did have it working two years ago when Mirko first posted instructions. 

I have searched through other issues and could not find anything similar. I am familiar with linux but not the BBB and have no idea how to debug this.

Any help would be appreciated.

Cheers,

Paddy

PaddyMcDonnell, 2019-05-31:
Hi, thanks for the response.

1. Yes, absolutely used the console images, tried these many times each following the guide exactly:
bone-debian-9.7-console-armhf-2019-02-03-1gb.img
bone-debian-9.8-console-armhf-2019-03-31-1gb.img
bone-debian-9.8-console-armhf-2019-04-07-1gb.img
bone-debian-9.9-console-armhf-2019-04-28-1gb.img
bone-debian-9.9-console-armhf-2019-05-12-1gb.img

Quite embarrassing really trying to get my boys interested in this stuff and then they sit watching me go through the same steps over and over without it working :) 

2. All testing done powered by LiPo

3. Servo rail on and GPIO 80 verified.

Cheers,

Paddy

PaddyMcDonnell, 2019-06-01:
Ok, I have plugged some servo's into the outputs rather than the motor controller and I can see that the servos twitch when the board boots and when I toggle GPIO from off to on, so I think the power side is fine.

When I lsmod the pru as I have seen in some other issues I get this:
`debian@beaglebone:/sys/class/gpio/gpio80$ lsmod | grep pru

uio_pruss              16384  0
uio                    16384  2 uio_pruss,uio_pdrv_genirq
`
Both lines have the same number, I have no idea what this means but I see that when you posted yours they have different numbers.

Ok, I spent more time going through previous issues and comparing with my setup.

I compared the /boot/uEnv.txt that Hagfelsh posted in issue 15 on the 30th of March and noticed that my uEnv.txt was different than his. He had dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo and I had dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo

I changed mine to match his and the servo output is now working. However I do not currently have a radio with me so I am doing this testing with a bluetooth game controller so, I am not sure if disabling dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo will affect this? But I have confirmed a number of times that having this enabled definitely prevents the servos from working. I will not be home for a week to be able to test this.

I have been doing every step in your guide, including steps 8 to 11 so I really do not know why it was different, but to be honest I am not familiar with sed and what you are actually doing there so it might be handy to include something like Hagfelsh had in his post in the guide, ie what uEnv.txt is supposed to look like (without all the comments) after making the required changes. It probably wouldn't hurt to actually explain what those changes do...if you have the time of course.

Anyway, thanks for the guide, without it I am sure this would have been much more difficult to set up. I recall last time I did it it a few years ago was not this easy.

Cheers,

Paddy

imfatant, 2019-06-01:
Re: dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo or /lib/firmware/BB-I2C1-00A0.dtbo

That's interesting. I didn't know that specifying UART4 would have this effect - i.e. that it would stop the servos working. Your suggestion of posting a working uEnv.txt is a good one. I'll update the guide shortly.

:)

closed 2019-06-01 by imfatant