imfatant/test #22: Readme Mistake?

Andrewiski, 2020-02-18:
sudo sed -i 's|#dtb_overlay=/lib/firmware/<file8>.dtbo|dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo\n#dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo\n#dtb_overlay=/lib/firmware/BB-ADC-00A0.dtbo|g' /boot/uEnv.txt

Did you really mean to commit out the UART4 and ADC overlays?  note the # in the sed command in the readme instructions

imfatant, 2020-02-28:
Yes, I did. I'm trying to remember the reason. I think it was because enabling them stopped something else working.. Now, if only I could remember.

Andrewiski, 2020-02-28:
I wonder if this is why my Battery is not detected. I will comment out the ADC and see if that makes the battery voltage detection work.  I am also struggling with the Spectrum DSMX receiver on the DSM port of the Beagle Bone Blue. If I load 3.6 Srducopter DSM works in 4.0 no DSM if I load 3.6 then 4.0 I get DSM.

Andrewiski, 2020-02-29:
Commenting them back out had no effect on DSM or Battery on BeagleBone Blue and Ardupilot V4.0.3rc1

Andrewiski, 2020-02-29:
Do you know if you can use the DSM labled port on the Beagle Bone Blue.  Reading this https://github.com/ArduPilot/ardupilot/issues/6641   and looking at 
https://github.com/ArduPilot/ardupilot/blob/05a0fe615b50fc2fa8dba11e541ba4f5740a1181/libraries/AP_HAL_Linux/HAL_Linux_Class.cpp

static RCInput_Multi rcinDriver{2, new RCInput_AioPRU, new RCInput_RCProtocol(NULL, "/dev/ttyO4")};

So it looks like DSM serial port 4 is used.

I see SBUS FD -1  115200 FD -1   written to the console

https://github.com/ArduPilot/ardupilot/blob/a153799e261de0fcc0008730e0f755e367f87d60/libraries/AP_HAL_Linux/RCInput_RCProtocol.cpp line 126

printf("SBUS FD %d  115200 FD %d\n", fd_inverted, fd_115200);

So not sure why no Spektrum DSM2 being detected with v 4.0

Andrewiski, 2020-03-02:
so issue seems to be that on bone versions of kernel /dev/ttyO4 is not avalible but on ti versions of kernel it is, next comes the issue of the a bug in the kernels so latest 19_ti kernel crashes. RobertCNelson did fix 19_bone kernel so it now works but no /dev/tty09 kernel.  I am able to run 9_ti kernel but now I get a bunch of DSM_DECODE_STATE_DESYNC messages and no DSM input.