imfatant/test #26: Using the 4 X Motor Drivers on BB Blue

ninjatrent, 2020-04-19:
Hello,
Do you know how to use the 4 x Motor Driver Outputs on the BeagleBone Blue with the Ardupilot image and Mission Planner?

I would like to deploy on a DFRobot Devistator Robot Chassis, a skid-steer type tank chassis. There is a 6V drive motor for each track. The onboard motor drivers perform great with the motors when testing with the libroboticscape motor test example. 

I'm having some difficulty locating the hardware address or pin numbers of the motor driver outputs on BB Blue and selecting these outputs in the Mission Planner app. There is an option in the app for skid steer steering with two drive motors but this setting assumes use of Pixhawk and similar autopilot boards.

silver2row, 2020-10-16:
Hello,

Compile like you do w/ ArduCopter but use ArduRover or plain, ole Rover. 

So...

    git checkout Rover-3.4

Then compile like usual.

Seth

ninjatrent, 2020-10-23:
Do you know if ardupilot will work with the Robotics Cape and the BeagleBone Black Wireless?

silver2row, 2020-11-20:
Hello,

I think it does. I think you can use ardupilot for the Robotics Cape, i.e. as the BBBlue was sort of referenced around it.

Seth

P.S. That is my perspective. Now, I have not tried the roboticsCape myself. There is another library too. librobotcontrol.